d-ronin/openlager #29: SPI support is needed

mlyle, 2016-04-25:


mlyle, 2016-04-27:
Probably should wait after the other performance bugs once we've wrung all the perf we can get out of async serial.

LitterBugs, 2017-06-21:
Curious if you have done any testing on SPI support?  Interested to use it for controllers/firmwares that don't support async serial > 250,000 baud.

mlyle, 2017-06-21:
There's no implementation, no.

Do any firmwares support streaming logging data out SPI?  (As opposed to talking to NOR flash chips, which is really something else)

LitterBugs, 2017-06-21:
*Flight supports SPIflash which is NOR chips.  So not that I know of.

mlyle, 2017-06-21:
Yah, I think it would be difficult to pretend to be a NOR flash chip, and if we did it we'd have a lot of the performance limitations that come with NOR flash.  The SPI pinning is there to allow if we ever need to go faster than 2.47mbps, but relatively few controllers have an exposed SPI and 2.47mbps seems to be plenty.

LitterBugs, 2017-06-21:
Unfortunately, *flight limits serial to 250,000.  Will have to see if they are willing to add support for faster rates.

LitterBugs, 2020-11-09:
Actually have a need to capture SPI data in the 24-32Mhz range.  Curious if this hardware would be capable of that speed if the firmware was modded?