d-ronin/openlager #54: improvement - flush & new file

LubinKerhuel, 2017-09-13:
First, let's say this logger is great and support much higher logging rate than its openlog counterpart. I've been looking for it for a very long time. Radio link to log were so-so, and openlog was too slow.

There are few pin which seems not used when using the UART connexion ; Is there any possibility to use one pin to force flush of current log and stop log when toggled to 0, and start a new file when toggled back to 1 ?
Currently, I added one wire which is doing it in a tiny hole nearby TP2 (might be a reset pin?). But that is certainly not the best way to do even if it works.

Another side remark: green led is too bright compared to red one. Red led is the one which provide useful information in the field, but I can hardly see it.

mlyle, 2017-09-13:
Thanks for the feedback.  --- can look into the possibility of doing this.

Where'd you get yours?  Note that I made a small batch myself, but all current production is just from Airbot who downloaded the design files and chose their own components/LEDs etc to put on.

LubinKerhuel, 2017-09-15:
I found only 2 place to buy it. I pick it at Airbot because of its slightly lower shipping rate for my place as I remember (France). I would pick the other one next time if it is the "original" one (yours?).

On performance feedback: setting the baudrate at 2 470 000, using 45% of the bandwidth (10.9 kbyte/s) for 15 minutes ~ half an hour. I get 0% error with old SD Cards formatted with either fat16 or fat32. That's all ok. (data recorded from a dsPIC)

kaspergrubbe, 2019-05-14:
@mlyle Is the one from RDQ from you or from Airbot https://www.readytoflyquads.com/high-rate-f4-data-logger-openlager ?

I would like to buy one, but would prefer to buy from dRonin.

tracernz, 2019-05-16:
@kaspergrubbe Buy it from wherever you can get it. The idea of making the design fully open from the start was that as many manufacturers as possible would pick it up and make the board easily available for everyone.