d-ronin/openlager #55: Help diagnosing if working?

dan-van, 2020-04-25:
Hi I've been trying to connect an OpenLager (purchased through Amazon.com) to a MATEKF405 running Betaflight 4.1.1.
No joy. I'm beginning to suspect maybe a problem with Betaflight, or my FC, but it would be nice to know OpenLager is working.
I bought a DigitalLogic analyser, but my FC doesn't seem to be trying to pull voltage low on the TX UARTS when they are set to Blackbox (ok, there is actually a single pulse on board startup, but nothing after that).
If I get a USB -> UART bridge, is it somehow possible to write data to the openlager unit from my laptop?

--- more detail on my betaflight setup ---
I get a zero-byte LOG000.TXT file each time I power up the flight controller.
I have betaflight set up to activate BB logging when armed.
Arming / disarming does not create a new file. Only powering the board creates a new file.
I've tried numerous different baud rates and 2 different TX pads on my FC (obviously updating the LAGER.CFG file).
Changing preallocGrow=true causes a 105mb log file to be written (assuming related to the preallocBytes : 104857600 setting). Obviously I can't parse this file.

Many thanks!

dan-van, 2020-04-27:
An update to this.
I got an FT232 and can send asci to the OpenLager.
The unit logs different length LOG000.TXT files depending on how many chars I type (using `screen /dev/...usbserial... <baud>`). The strange thing is that the file is just full of 0x00 values, no asci chars (I was expecting it to just capture the data on the TX line).
Maybe I've got some other config problems or somehow mismatched baud rates, but should I expect to type asci in a terminal and capture the same bytestream on the openlager?

tracernz, 2020-04-27:
> should I expect to type asci in a terminal and capture the same bytestream on the openlager?

Yes, that should be what happens. The most likely scenario is one or more of the parameters is mismatched (baud rate, parity, stop bits), most likely baud rate. The openlager defaults to 115200 baud, but can be overridden with the config file (lager.cfg on the SD card, which will be created from defaults if it doesn't exist). The other parameters cannot be changed on the openlager.

dan-van, 2020-04-27:
Many thanks, I'll use something more controllable than 'screen' for partiy bits etc. 
Closing the question: Having an FT232RL seems like the best way to verify operation (I guess technically there could be a self-test mode that relies on TX and RX being bridged on the OpenLager, but seems like poor effort / reward at this stage).

In case it helps anyone else (some other things that had me puzzled for a while):
Betaflight (4.1.1) with `blackbox_mode=NORMAL` doesn't start emitting data until the quad is fully armed. i.e. just configuring Blackbox to start in line with your 'arm' switch from your transmitter is not sufficient. The board needs to pass all pre-arm checks (i.e. go to full 'props spinning', and not have USB attached!).

Setting blackbox_mode=ALWAYS does emit UART data from power-on (seen on logic probe), but it's not really designed to work (e.g. missing headers are probably expected) and at time of writing the OpenLager didn't actually flush any data. I've probably messed up baud rates. Once I've verified capturing accurate ascii data from my laptop I'll go back to that.

Many thanks for the project!

closed 2020-04-27 by dan-van