d-ronin/openlager #58: Problem with Openlager

rs3hc, 2022-04-10:
Sorry for posting it as a issue here but i cant find any detailed informations on the web.

i got a openlager from a buddy for logging a Betafpv Toothpick F722 2-6S AIO Brushless Flight Controller 35A(BLHeli_S) which dont offer storage...

I soldered 5V, GND and TX6 from FC to RX on openlager.
In Betafligth i configured UART6 as Blackbox logger ans 2000000 baud, ans Serials Logger, 2khz and Gyro_Scaled in the Blackbox section.
Should be ok...

I tried multipe SD Cards from Class4 over U1 32GB to U3 64GB A2 V30.... also tried all formats FAT/FAT32
Nothing works...

If i Power the FC the green LED on Openlager lit up correctly but the red LED (should be for write process?) never lit up.

Can anybody discripe how it should look like if it will work?

maciejlagowski, 2022-04-10:
same issue here, tried different SD cards, different file formats etc..

Zuldan, 2022-04-11:
I don’t know the answer to your issues but if you guys also mention which Openlager chip you are using it may help in the diagnosis. As far as I know there is an Airbot and JHEMCU version, possibly others exist…

maciejlagowski, 2022-04-11:
Mine is Airbot Openlager

rs3hc, 2022-04-11:
[image]

maciejlagowski, 2022-04-14:
My problem was solved by inserting a card with the lager.bin file to flash. Surprisingly now the device works :)

rs3hc, 2022-04-15:
> My problem was solved by inserting a card with the lager.bin file to flash. Surprisingly now the device works :)

Only "lager.bin"?
How long did it take?
What did the red LED while updating?

For what do we need these two other files? (bootlager.bin, ef_lager.bin)

maciejlagowski, 2022-04-15:
> > My problem was solved by inserting a card with the lager.bin file to flash. Surprisingly now the device works :)
> 
> Only "lager.bin"? How long did it take? What did the red LED while updating?
> 
> For what do we need these two other files? (bootlager.bin, ef_lager.bin)

I put only lager.bin file on the SD, red LED was flashing for about 20 seconds maybe, then I powered it off and checked the SD card - there was config and log file.

rs3hc, 2022-04-15:
Oh OK..

I tried 3 sd cards with lager.bin and nothing happened. Only green Led was on

rs3hc, 2022-05-02:
Any one can help??

closed 2023-07-12 by mlyle