d-ronin/openlager #61: Witch firmware shold I flash?

sanmaodeai, 2023-05-19:
Hello guys, I make a openlarger hardware ,but I don't know witch firmware should I flash ?
bootlager.bin ,  ef_lager.bin  , lager.bin
witch one should I flash?
I want use this hardware for  the BF's blackbox recoder . Use serial to communication.

mlyle, 2023-05-19:
ef_lager.bin is the entire flash image, including the bootloader (bootlager.bin) & the firmware (lager.bin).

The update process uses a lager.bin.

closed 2023-05-19 by mlyle

sanmaodeai, 2023-05-19:
> ef_lager.bin is the entire flash image, including the bootloader (bootlager.bin) & the firmware (lager.bin).
> 
> The update process uses a lager.bin.

thanks for your reply.And if I flash the  ef_lager.bin , the start address is 0x8000000?