d-ronin/openlager #62: Openlager blinking RED LED and no files

rogerfpv1989, 2023-07-08:
Hi,
I just bought and connected an Openlager to my FETTEC G4-N FC. Upon applying power, grenn LED goes ON and RED LED goes on, blinking with a certain pattern. However no files are created on SD Card.
The RED LED blinking pattern stays ON forever (until power removal that is); it is present whether there is an SD Card installed or not; I've tried the recommended SD Card, and other SD cards (to up to 64GB) but nothing works. I've tried putting the lager.bin file on the SD cards, but nothing changes, same behaviour. It's like the Openlager is not reading the SD cards...any ideas? Need help!
Thaks in advance

mlyle, 2023-07-08:
Sounds like you probably have a defective board.  Talk to your vendor: we don't provide support.

The blinking pattern is morse code that will tell you the error, if that helps.

closed 2023-07-08 by mlyle

rogerfpv1989, 2023-07-12:
Thanks for the help.
So I am no morse expert, but with my phone camera (red LED is not easy to see right next to the very bright green LED) and a morse code translator, here's what I managed to decypher: DCARD+NC+DCARD+NC+ etc
What is this supposed to mean?

mlyle, 2023-07-12:
'CARD' indicates that it never received a reply from the SD card.  https://github.com/d-ronin/openlager/blob/master/src/openlager.c#L594C1-L597C10

rogerfpv89, 2023-07-12:
Alright, thanks. Can you tell me if there is a maximum size of Sd caed rho Openlager is able to handle? 16GB? 32GB? 64GB?

mlyle, 2023-07-12:
Oh, that's a good question.  It looks like we only have code to support SDHC, so that would be 32GB.

I don't believe it would fail with a 'CARD' error on larger cards, though.

mlyle, 2023-07-12:
Nevermind: SDHC and SDXC protocol are identical from our point of view; I'd expect larger cards to work (as long as they have a FAT32 filesystem and not exFAT).

I will try to try some large cards soon.  I'm not sure I ever used over 16GB.  In any case, this wouldn't explain your problem.